Domesday86/SmallyMouse2 #6: Amiga 1200 doesn't detect right mouse button press

juwivive, 2017-11-30:
I'm trying to get the SM2 to work with an Amiga 1200 (revision 2B motherboard).  The interface does not detect a right mouse button click (left button and movement are fine).  Using the same setup on an Atari ST (swapping the X2/Y2 over) works without problem.

Using a multimeter, I tested the change in voltage that happens when a mouse button is pressed.  For the left button, the reading goes all the day down to 0V.  For the right button, it goes to 0.3V and hovers around there.  If I disconnect the interface from the computer, the voltage does go to 0V when the button is pressed.

It seems that certain Amigas have this problem with other USB mouse adapters:

http://www.amibay.com/showthread.php?53905-Amiga-1200-USB-Mouse-adapter-No-Right-Mouse-button

Certain adapters claim to solve the issue (though it's not stated how):

https://www.ebay.co.uk/itm/MKIV-Amiga-USB-mouse-adapter-RIGHT-MOUSE-BUTTON-FIX/121715285463

Some people suggest that you need an external transistor somewhere:

http://eab.abime.net/showthread.php?t=64460

Is there a fix that can be made (e.g. by building a small circuit sitting between SM2 and computer) that will resolve this?

simoninns, 2017-11-30:
That's odd (very technical response, I know :) )...  Well, since the microcontroller is pulling up the lines to 5Vs it is most likely caused by the level not dropping all the way to 0V when the button is 'pressed' - this could only be caused by a very poor design in the host computer itself.  Simply placing a 4K7 resistor between the button signal and ground should cure the problem (4.7Kohms should be high enough to do the trick without 'straining' the internal pull-up).

If the BBC micro header is unpopulated, you could place the resistor there; otherwise you could solder it to the underside of the board (make sure the leads are insulated and don't cause any shorts).

Can you try this and see if it works?  If so please report back so I can update the project information and, thanks for reporting this!

Septercius, 2017-11-30:
Thanks for the response!  I'm currently using the 10-pin header, not the Beeb header.  The 5V pin and the 0V next to it are connected to a 5V PSU (2A, I think).  The rest of the pins are connected to a cable that leads to the standard 9-pin port on the back of the computer.  Just to be absolutely sure: I should leave this as it is and place the resistor in parallel, across the RB and 0V lines?  I've attached (I hope) a quick mock-up schematic that illustrates what I think I need to do.  Could you confirm I'm on the right track, please?  (I'd prefer not to blow anything up!).

[image]

simoninns, 2017-11-30:
Exactly right.  The same pins (RB and 0V) are exposed both on the 10-pin header and the 2 row IDC header for the BBC micro; if you simply trace the connections (using your multimeter) you could simply hook the resistor up using the holes for the IDC (but how you connect it isn't important, as long as it's between the RB and 0Vs) - it was just a suggestion.

Septercius, 2017-12-01:
Unfortunately wiring up the 4.7K resistor across the RB and 0V pins didn't make any difference.  The voltages were still 5V (button released) and 0.3V (button pressed).  I checked the circuit by touching both legs of the resistor together (I'd wired it in using some probes) and the computer acted as if the button was held down.

I also tried the next size up of resistor that I could find - 10K: same result.

simoninns, 2017-12-01:
Is there an accurate schematic of the motherboard for the version of Amiga you have?  If you could point me to it perhaps I can figure out what's going on.  There has to be a reason for it not working like this and, if I can figure that out, I could most probably suggest a better workaround.

Septercius, 2017-12-01:
There's some here:

http://www.amigawiki.de/doku.php?id=en:service:schematics

My motherboard is a revision 2B, so presumably it's version 2 of the schematic.

simoninns, 2017-12-02:
Pin 9 of the DSUB (RB) is connected to P0Y/P1Y of the 'PAULA' chip.  Since these are market 'POT0Y and POT1Y' it's safe to assume the chip is acting as an ADC.  The 'middle' button is connected the same way (although I would presume this isn't in use - but it would be failing in the same way).  All of the other lines are connected via a 74LS166 (an 8-bit shift register) which explains the difference.

The software must be expecting the ADC to read a very low value which is basically a bug in the mouse driver...  it should follow the TTL rules (anything below 0.8V should be considered a zero).  

I'll have a look at how to solve this and post back.

simoninns, 2017-12-02:
The best thing I can think of (and I have no way to test as I don't have any Amigas) is to use a non-inverting voltage follower on the RB pin.  You can form this circuit with 2 resistors and a transistor (I think a normal small signal NPN will do it, but a MOSFET would provide more range).  If you could try the following circuit it should solve the issue:

[image]

Septercius, 2017-12-02:
I'll give it a whirl and see what happens, though I'll need to order some bits first.

Looking at the RS web site, there are lots of different BC547 transistors out there:

https://uk.rs-online.com/web/c/semiconductors/discrete-semiconductors/bipolar-transistors/?searchTerm=bc547

Does it matter which one I use?  The main difference I can spot (other than physical size) is in "Minimum DC Current Gain" (whatever that means).

simoninns, 2017-12-02:
The transistor probably doesn't matter too much as long as it's NPN.  I'd go for the 547B, although if you have a BC337 handy, it would probably work too.  Although the configuration is a 'voltage follower' the input to the base of the transistor is either 0V or 5V, so it's just swinging from min to max, so the details of the transistors finer qualities shouldn't make too much difference.

Basically the circuit should work like this:

If RB_in is 0V power cannot flow from the collector (pin 1) to the emitter (pin 3) of the transistor so the circuit takes the path of least resistance to ground.  If RB_out is 5V then power flows through the transistor making RB_out 5Vs.

The 10K resistor acts to limit the current draw from the microcontroller and the 4K7 pulls RB_out to 0V when the transistor is 'closed'.

The downside of a transistor is that it's not a 'perfect' switch, so it will never be truly 5V or 0Vs... the better the transistor - the closer it gets to being a switch (with MOSFET transistors being best at this).  Since the resistor is on the ground-side of the transistor it should present a much better 'ground' to the Amiga's ADC than the microcontroller pin... Thereby solving the issue...  (I hope! :) )

Septercius, 2017-12-16:
I finally found time to try the suggested circuit.  The 5V and 0V lines are connected to a separate 5V PSU.

The voltage is 4.44V if the right mouse button is pressed; if I press the button, it reduces to 3.85V, not 0V.  I have checked my circuit and can't see anything amiss.  If I connect everything as it was before (i.e. without the circuit), I see the previous behaviour (5V on not pressed, 0.3V on pressed) and all the other mouse functions work correctly.

simoninns, 2017-12-16:
Can you try increasing the value of the 4K7 resistor (10K, 100K, 1M) and see what effect it has?  If nothing works let me know and I'll try to find some time to set it up on the breadboard and look at what's going on.

snnsyhn, 2019-09-06:
Thank you.
I replaced the PAULA and the right mouse button problem solved.

simoninns, 2019-09-06:
Sounds like the root cause of this issue is identified; and the SM2 is functioning correctly.  Thanks for the feedback, I'll close this issue for now then,

closed 2019-09-06 by simoninns

gianlucarenzi, 2019-09-06:
I changed PAULA too, and it worked for me. As soon the board was in the customer's side, the problem persists. I had no time to check it out again... sorry. But I suppose something weird who breaks PAULA from the mouse port. I dunno.

ChuckyGang, 2020-01-31:
https://www.tsb.space/knowledge-base/amiga-1200-right-mousebutton-does-not-work/

JaneDoe649, 2020-07-19:
I had the same issue: the right mouse button did not respond when using the Smally Mouse. Apparently Paula's internal pull up resistor has a very low value which induces a current of almost 20mA when the level on Pot0X is set at 0V. Apparently the microcontroller cannot absorb it which causes too much voltage at the input of Paula to detect level 0.

I solved the problem by connecting the PC6 and PC7 pins together and modifying the declaration of the right button like this: #define RB ((1 << 6) | (1 << 7))

When the right button is pressed the two lines PC6 and PC7 drain the current and level 0 is well detected by Paula.