Domesday86/SmallyMouse2 #7: Atari ST some problems

pazuzu72, 2019-02-24:
Hello everyone, on the **Amiga** it works very well (**500**, **600**, **1200**). On **Atari 1040 STE** (the only one I have available) has some problems. If I move the mouse too fast, it does not seem to be behind the speed and resets the pointer arrow position. Even setting the jumper on **slow**. 

Can someone confirm or not the same problem?

dh219, 2019-12-06:
Yes, I can confirm. The ST range seems particularly sensitive to too high stimulus -- I can reproduce a similar effect on my 90s ball mouse by spinning the rollers with my finger.

The same problem is very hard to replicate on my Archie A3000, so I think it might be an Atari phenomenon and might explain why my TOM2 has such a low max speed in ST mode.

I'll try to have a dig in the firmware and see if I can improve it a bit.

dh219, 2019-12-31:
I've made some changes to my fork of the firmware that I think overcomes this issue -- firstly it looks like the Atari range can't handle polling rates over about 500Hz. Setting the rate limit to 500 solves the jumping cursor issue.

Also I've added a DPI scaler parameter that will basically scale down the effective DPI of the mouse as modern mice are far too high by default. I find a setting of 2 or 3 is good for modern 'slow' mice. 4 is about right for faster ones.

If you have the means to recompile and flash the firmware, have a look on my fork:- https://github.com/dh219/SmallyMouse2

pazuzu72, 2020-01-05:
Great! I have tried your version. I set the variable to 3 and it works great on 8 bit atari too. It is also better on amiga. Thank you :-)

simoninns, 2020-02-08:
In the V1.2 release I have added an additional 'jumper' configuration to turn on DPI divider support (which, by default divides by 2).  I have also lowered the quadrature limit when the SLOW header is used.  The DPI divider jumper should be placed between D7 and 0V on the expansion header which was previously a general purpose GPIO pin.

closed 2020-02-08 by simoninns