Domesday86/SmallyMouse2 #8: At least RMB need to be open collector output

TzOk83, 2019-07-25:
Actually all outputs should be open-collector type, but on Atari ST the right mouse button is a must, as in other case it will conflict with a joystick fire button (they are shorted together on Atari side).

simoninns, 2019-07-26:
Can you elaborate more about the use-case here as I'm not really sure what your issue description is trying to solve.  This can't be a 'must' since there are many people using SM2 with Atari STs and it works.  What is the actual issue you are testing in order to reach your conclusion?

TzOk83, 2019-07-30:
This is a formal requirement of a mouse/joystick port - all outputs should be open-collector type (either in high impedance, or shorted to ground). In most cases it will work with a normal output, but problem will arise when you will try to use mouse on port 0 and joystick on port 1. Right mouse button on port 0 (pin 9) is connected together with a fire button on port 1 (pin 6). All signals are active low, so at idle they are held high (+5V). When you press fire on the joystick, the pin 9 of port 0 is directly shorted to ground, while on the same time SM2 tries to hold this pin high. In most cases it will work, but stressing the PC6 pin over its limits, and unnecessarily drawing over 20mA. Microswitch based joysticks will work fine, but some universal Atari/CPC or game-pads with rubber switches - won't due to high closed switch resistance.

dh219, 2019-12-06:
Can confirm what TzOk83 says. At least on the ST pressing fire on joystick 1 causes a short in the SmallyMouse2 with stock firmware. I've instigated a fix (for buttons only) in my fork. I'm not really familiar enough with Git to know how to submit a change to your repo for that file only. The diff is small and visible here, however, if you're interested: https://github.com/dh219/SmallyMouse2/commit/803f7c1757fec1a24074387a8cf2013e2163f189

Regards.

TzOk83, 2019-12-07:
@dh219 You should have created a Pull Request to the Master branch of the original project in order to integrate your patch with it.

simoninns, 2020-02-08:
In the V1.2 release I have added support for open-collector mouse buttons for all 3 buttons (as it shouldn't have any negative effects on other machines) - I've tested it with a BBC Micro and an Acorn Archimedes and it functions fine.

closed 2020-02-08 by simoninns