Domesday86/SmallyMouse2 #9: centuriontech.eu "usb mouse module"

blark, 2019-10-21:
Just wanted to let you know, in case you haven't seen it, that this company is selling a version of your product: 

https://centuriontech.eu/product/a1200-usb-mouse-module/

While they have linked back to your GitHub repo they have not released the design source files as per the project license.

blark, 2020-01-25:
Also, I made my own for the A1200 Rev 2B motherboard. You can find my fork [here](https://github.com/blark/SmallyMouse2) if you want to make one.

simoninns, 2020-01-25:
Nice :)  I'm aware of the centurion tech product - it's hard to strictly enforce things; but you seem to have solved the issue in the right way!

simoninns, 2020-02-08:
Added a 'reminder' in the V1.2 release.

closed 2020-02-08 by simoninns