Domesday86/SmallyMouse2 #12: When moving mouse very slowly point stops moving.

xwildx, 2020-06-13:
Standard SmallyMouse2 Board 
V1.2 Firmware
Amiga 500 Rev5 motherboard
When moving the mouse very slowly the point stops moving. If you speed up mouse movement a little pointer moves normally. This is only a small issue. but the problem becomes much worse when the right mouse button is pushed. Making it difficult to select menu items.

xwildx, 2020-06-14:
Did some more testing and found the following:

Commenting out the following 2 lines in the code.
mouseEncoderPhaseX = 0
mouseEncoderPhaseY = 0

Fixes the problem of the mouse stopping while moving very slowly. Of course this reintroduces the directional lag issue reported by dh219. I find this a much less annoying issue. 

As for the RB issue. This looks to be something more problematic. I rewired  the MB to pin 9 (RB) input on the Amiga. So I'm using the MB on the mouse as the RB on the Amiga. (ie I swapped Pin 6 and 7 on the universal output connector).

So the MB acts like the RB, as far as the Amiga is concerned. This works normally and mouse movement is not affected when the MB (acting as RB) is pushed and held.
But if I push and hold the actual RB the mouse movement is still affected. Even if I disconnect the RB output from the amiga.

So it's not a problem with my Amiga hardware.

I have also tried a number of different mice. Both wired and wireless. All exhibit the same behaviour.

Maybe a problem with the USB library??

xwildx, 2020-06-18:
I have changed some lines in the code so the MB now outputs on pin 6 port c on the AVR and the RB now outputs on pin 5 of port c.

So now the MB is output to the right mouse button input on the Amiga
and the RB is output to the middle mouse button input on the Amiga
(Basically the same thing I did above but this time i did it in software not hardware)

If the MB is pushed and held it acts as the RB (menu) on the amiga and works fine.
If the RB is pushed and held it acts as the MB (not really used) on the amiga and the mouse movement is affected.

So I looks like the problem is only with the RB.

Could someone confirm this?? If you push and hold the RB on the mouse and slowly move the mouse  at the same time does it make the mouse movement erratic?

I have built 2 SmallyMouse2 boards and both exhibit this behaviour.

simoninns, 2020-06-18:
Thanks for reporting the issue; I don't have an Amiga, but I will test this out once I have the opportunity and see if I can recreate the observed behaviour.

xwildx, 2020-06-19:
Thanks for looking into it. I have done a bit more testing. 
I powered the SM2 from a 5V supply plugged in a mouse and connected an oscilloscope to the Y2 output on the universal output connector. Note the SM2 is not connected to my Amiga.

I then tried to move the mouse, as steadily as possible, in the "UP" direction and observed the waveform. While moving the mouse I then pushed the RB.

As you can imagine moving the mouse steadily enough to get a "clean" waveform is difficult. But when I push the RB I can see the waveform change.

From this
_____         _____
|         |        |         |
|         |____|         |____

To This
__                __
|  |                |  |
|  |________|  |________

If the pulse is at  +5V say for 20ms and at 0V for 20ms, when the RB is pushed it changes to about 5ms and 35ms.
So the pulse width is narrowing with the RB pushed but the frequency stays the same.
And the slower you go the narrower the pulses get. But only with the RB pushed. The LB and MB don't have this  problem.
I will try to film this and put a video on youtube over the weekend.
Will post a link when its up.

xwildx, 2020-06-21:
I have put up a short video showing a capture of my oscilloscope. It's a bit hard to see but at about 6 sec I push and hold the RB and you can see the pulses become shorter. Some  are down to about 5ms or so.  This is only one output but it's the same on all 4 outputs. If you pause the video at about 8.5 Sec you can see the short pulses.

At the end of the video there is a small clip showing the mouse movement. About half way up the screen I push the RB and the mouse stops moving.

https://youtu.be/6IHmSAWSCrc

If there is anything else I can so to help let me know.

simoninns, 2020-06-22:
I've confirmed that the right button issue is reproducible and I get the same issue as observed on my test board.

It looks like the slow movement and RB held issues are possibly unrelated (so should be two issue reports), but I'll leave them as one for now.

simoninns, 2020-07-30:
I've pushed a fix which I believe solves the observed problems; can you compile and flash the latest git commit and test?  Please let me know the result @xwildx

xwildx, 2020-08-01:
Flashed the latest commit and have been testing it. Can confirm the slow movement and RB held problems are fixed. Thank you for the effort.

simoninns, 2020-08-01:
Great, I'll put up a new release shortly then.  Thanks for the assistance with testing!

closed 2020-08-01 by simoninns