Domesday86/SmallyMouse2 #23: fix for A1200 Rev 2B Amiga RMB

kipper2k, 2021-06-10:
After building and testing the smally mouse on Amigas i found that the Rev2B A1200 Amiga RMB did not work. the problem was identified as bad caps and non working Paula. There is someone on Ebay who is selling "fixed" adapters but as far as i can tell he has not released any details on how he done it.

 I believe i have fixed the problem by placing 2 mosfets in cascade from the chip to the RMB input. I will play around with the resistor values and do a little more testing and then i will release the mod here so all can benefit

Sgw32, 2022-10-12:
@kipper2k Thanks for the idea!
Does it work on Amiga 500?

kipper2k, 2022-10-12:
yah, works on all

kipper2k, 2023-07-18:
Forgot to add this, it is a render of how it looks on the motherboard.
There is a USB header so people can run a USB cable if they don't want to
use the actual USB plug,. below the Smally is my video Hybrid board which
you break off and an expansion board for IDE so people can mount it
wherever :)

Brian

closed 2023-11-20 by simoninns