Domesday86/SmallyMouse2 #25: Use of Smally mouse on an Amiga motherboard

kipper2000, 2023-07-12:
Hi, I am designing an Amiga 500 motherboard and i would like to add the layout of your smally mouse to the motherboard. I have someone screaming at me that i am not allowed to do it. I will not add the chip for it, just the layout and am making the motherboard gerber free to use so people can do thir own boards,

 Can i have permission to add the schematic to the motherboard. A USB mouse is much better than the already mounted serial mouse.

Thanks

kipper2k

simoninns, 2023-07-17:
This is allowed as the board design is CC licensed - you can read about this in the README.md in the main repo.  Provided you are intending on releasing under the same share-alike license, there is nothing to discuss :)

The software is GPL, but as this is only on the AVR itself, it doesn't knock-on to anything else that might be in your project.

If you are not intending on publishing your schematics and board designs under an open license - then it's not allowed.  However, if this is just because you don't want to use sharealike CC and want to use another open license - just let me know what you are planning and I can provide written permission.

(sorry for slow response, I'm on vacation at the moment :) )

kipper2000, 2023-07-18:
Hi Simon,

Hope your holiday  is going well :). Thanks for the reply.  The overall
scope of my project is to design a  new SMD Amiga 500+ motherboard. I built
the first test board back in early 2021 and then priorities shiftd so it
has been shelved ever since.  The current version motherboard itself has
been tested with onboard Gotek, 2MB chip mem and a quadrom option that
allows a user to toggle up to 4 x 512MB Rom images using an Amiga warm
reset (remembering current selected rom on next reboot. There is no issue
using these addons..  The Smally board circuit is on the motherboard also,
(see picture)  I still need to verify the circuit.  1 question for the
outputs to the original mouse port, do i need to put diodes on to protect
the chip in the event someone uses the original mouse port ?

 I am also adding 8MB fastmem and IDE functions so people can use CF card,
SD, and HDD using one of my old designs (based on NedoPC way back in 2015).
I  have found a way to implement 1 of those cheap Chinese wifi/bluetooth
Sony PS2 joysticks. Basically I am trying to give the A500+ all the options
without all the plugins.

 My desire for this board is to release the Gerbers free, and allow people
to build their own. This way they can do the work and I am assuming it
would be considered personal use. I know that people will probably want to
build a few themselves at a profit but if you check ebay etc it is
happening with boards etc that are not supposed to be commercially sold.

I am including your updated Smally schematic that I did with Mike Stirling
(he was the real brain :). It fixes the RMB problem on one of the A1200
version motherboards. I removed the header for the Atari version as
obviously no need on this motherboard

 I will of course give you full credit shown on the motherboard with
description of your choice. and like i said it will be up to the end user
to supply/program the chip. Ultimately I would like to just have the Smally
circuit on the motherboard so people have the choice of whether to use the
serial port or USB.

Thanks for your time, let me know if you need more info and if you are ok
with what I would like to do.

Brian, (kipper2k)

simoninns, 2023-07-27:
I would prefer that you followed the CC ShareAlike license rules... the idea is that, just as you take my design and benefit from using it in your design, others can take your design and benefit from using it in theirs.  What you are proposing is freely using and extending my work into something which you then provide closed (gerbers don't count as they are not really possible to manipulate).

Provided you are not-for-profit; I will be flexible and state you have permission to incorporate the schematic from SmallyMouse2 - but please consider using a sharealike design so others can benefit from your work as you are from mine :)

closed 2023-07-27 by simoninns